SpiNNakerManchester/spinnaker_tools #3: Documentation links broken

mossblaser, 2015-04-23:
The documentation linked to in the README (following #2 ) has now been moved in the wiki and so the links are now broken.

Is there some way to generate permanent links to the documentation which will survive future reshuffles (especially considering wiki page names now contain the software version number *and* hand-written section numbering, changing either of which can break links)?

alan-stokes, 2015-04-24:
Hi Johnathon. 

i've changed the link now. But theres no way to make perment links, as theres constant debate on where these user pages are going. 

This was the reason why having a seperate one in each repos was a dubious idea, as its not easy to renemebr that the spinnaker tools has one which is needed to be adjusted as well. and this one is tied to a speciific release and is in the source code of the repos. Now youve got a mixture of docuemntation and version controlled source code. 

Ill keep this issue open, as there needs to be a discussion on keepign this file with linsk in the first place, due to this mismatch.

mossblaser, 2015-04-24:
> i've changed the link now.

Thanks for that! A random GitHub tip by the way: by including a mention of issue numbers (e.g. #3 in this case) in commit messages and pull requests, people subscribed to the issue will get notified of progress. The commit also appears in the issue history. Can be quite handy sometimes!

> But theres no way to make perment links,

I should probably correct myself here; what I really mean is that there is a need for *stable* links, i.e. ones which can be expected to remain valid for a very long time. This is absolutely critical for large (and useful!) pieces of documentation such as this, especially when the documentation is public-facing. Example use cases where reliable links are required, unrelated to the README example:

* Users will be relying on their own bookmarks to reference useful information they care about
* Third-party projects will want to point out relevant documentation to their users or people wishing to reproduce results.
* Any large document will make use of cross-references, and in this day-and-age this means links.

> Now youve got a mixture of docuemntation and version controlled source code.

This raises a related issue when it comes to documentation: documentation is subject to many of the same challenges as source code when it comes to maintaining historical versions. It is very useful to be able to get access to snapshots of old documentation and source code from roughly the same time period: for example this may be critical when attempting to reproduce experimental results for "quick one-off-hack" projects which only work on old versions of the toolchain.

As such, as well as the need for stable links to exist, there is also a need for deep/permanent links to specific versions of documentation. This is already possible for code in the repository: one can refer to git commits by their hash in the worst case. GitHub also provide such facilities for their wikis but unfortunately these links are obliterated when a page changes its name. This means that by renaming the wiki page (and making copies) on each release does not work here, no matter how consistent the naming scheme.

> Ill keep this issue open, as there needs to be a discussion on keepign this file with linsk in the first place, due to this mismatch.

Solving the issue of stable links should instantly make this a non issue.

alan-stokes, 2015-04-24:
Hi Johnathon, 

thanks for the tip. didnt know.

the last few days were moving us to this state of persistent wiki pages pointing at the correct tags in the soruce code etc. Please refer to the new look of our wiki pages. we're currently yet again, reinventing it to work with some html pages section instead (whilst keeping the markdown) so that we can control the look more precisely).

keeping this readme.md in the source code itself though, still sits wrongly with me, as this then needs to be adjusted each time a relase is done, in the soruce code, and not in the documentation area. Id rather there not be any doc in the code, or maybe (as rowley pointed out to me) we could have the readme.md point at the home page directly, and say "further info is lcoated here") and not directly to a spefici page that will cahnge on a relase by release basis.

qian-liu, 2015-04-24:
The example link is not the latest. Thus I got some error when running the retina_example.py
https://github.com/SpiNNakerManchester/SpiNNakerManchester.github.io/wiki/2015.004%3a-Little-Rascal-%3a-0.2-PyNN-on-SpiNNaker-Installation-Guide#Examples

I believe this link points to the latest example:
https://github.com/SpiNNakerManchester/SpiNNakerManchester.github.io/wiki/2015.003%3a-Just-Testing-%3a-0.2-PyNN-on-SpiNNaker-Installation-Guide#Examples

I don't find a better place to notify this, sorry to write it here.

alan-stokes, 2015-08-26:
these links seem to work now. so closing issue. Ive also added a few readme.md's to some of the modules.

closed 2015-08-26 by alan-stokes