SpiNNakerManchester/spinnaker_tools #8: spinnaker_tools/setup can assign wrong SPINN_DIRS value

ericnichols, 2016-03-11:
The following line of code:
[https://github.com/SpiNNakerManchester/spinnaker_tools/blob/master/setup#L7](url)
assigns the current directory to SPINN_DIRS. That works well when compiling from within spinnaker_tools, but... 

The current directory might not be (as was the case for me) the spinnaker_tools directory when compiling multiple repositories' code with a single script. The result is SPINN_DIRS being set to the wrong location, causing:
/home/eric/Code/git/sPyNNaker/neural_modelling/make_lib/Makefile.SpiNNFrontEndCommon: No such file or directory

SOLUTION:
I got this solution from:
[http://stackoverflow.com/questions/18136918/how-to-get-current-relative-directory-of-your-makefile](url)
I tested it and it works perfectly.

Replace the line:
[https://github.com/SpiNNakerManchester/spinnaker_tools/blob/master/setup#L7](url)
with:
mkfile_path := $(abspath $(lastword $(MAKEFILE_LIST)))
SPINN_DIRS := $(notdir $(patsubst %/,%,$(dir $(mkfile_path))))

The solution will work regardless of the compile location.

-Eric.

mossblaser, 2016-03-11:
This is a good point.

Looking at the Stack Overflow page you refer to, it appears the snippet you suggest is intended for use in a makefile rather than a shell script. Sadly, it does not work for me either way. Might I offer up the following alternative substitution for [line 7 of the setup script](https://github.com/SpiNNakerManchester/spinnaker_tools/blob/master/setup#L7)?

`SPINN_DIRS="$(readlink -e "$(dirname "${BASH_SOURCE[0]}")")"`

@ericnichols, would you try this out in your copy and verify that it works for you too?

In case you're interested, to break this down:

In a normal script, `$0` would get you a relative path to the current script. Unfortunately since the `setup` file is sourced and not executed, this technique does not work. As a workaround I instead use the `BASH_SOURCE` array variable is usually used alongside `FUNCNAME` array variable to produce stack-traces of Bash scripts. In this instance, since the line in question is not part of a function, `${BASH_SOURCE[0]}` actually gives the name of the file being sourced (which is what we want!). The Bash man-page describes these variables more fully.

Next, `dirname` is used to turn the (relative) path of the directory containing the `setup` script: the `spinnaker_tools` directory.

Finally `readlink -e` is used to resolve the full, absolute path to the `spinnaker_tools`. This step is important since changing directories would break a relative path.

ericnichols, 2016-03-11:
I tested your fix substituting on line 7 of the setup script:
SPINN_DIRS="$(readlink -e "$(dirname "${BASH_SOURCE[0]}")")"

on my Fedora 23 machine and it works great!

Thanks a lot!
Eric

mossblaser, 2016-03-11:
@rowleya are you in charge of this repository at the moment? Would you be prepared to apply this patch? Thanks! :)

AlexRast, 2016-03-11:
I think great discretion needs to be applied when using 
auto-detecting/auto-configuring lines in makefiles (or scripts, or anything 
else). It's too easy to end up with 'magic' code snippets that someone reading 
years later doesn't know what's going on. And then something that years ago was 
relevant but is now completely superseded or redundant ends up surviving as a 
zombie, because nobody dares remove what might be doing something critical. It's 
even easier (far, far easier, in fact) to lose track of which precise directory 
is being used to compile what, and then when you have multiple possible 
directory sources working out why a given compile gives crashing code (due to 
compiling or linking old source or a different version or a thousand other 
variants of I-got-the-wrong-directory) can become a real headache.

So I think this should be given more thought, in the context in fact of a 
careful review of the entire makefile structure, before being applied. We need 
to make sure we have a solution that will work for *everyone*, and that won't 
lead to confusion later.

mossblaser, 2016-05-18:
This issue has been fixed and will make it into production upon the conclusion of the great SC&MP/sark merge of 2016(!)

closed 2016-05-18 by mossblaser