SpiNNakerManchester/spinnaker_tools #23: spin1_flush_dma_queue function needed in spin1_api

AlexRast, 2016-07-21:
Sort of self-evident what is wanted here: a function to dump any pending DMA transfers. 

The PRAERIE I/O interface uses DMA to buffer incoming messages, but there is also a PRAERIE command: RESET, for the interface. If RESET is applied while there are pending DMAs this can create unexpected conditions, and existing TIDs become invalid. RESET should first flush the DMA queue then reinitialise the interface. 

Similar functionality could be needed in any application where absolute receipt of all DMA transfers (triggered by input events) is less critical than ensuring the simulation doesn't grind to a halt

lplana, 2016-07-21:
Some initial thoughts. The function should, at least:
- [ ] flush the software DMA queue,
- [ ] flush the hardware queue in the DMA controller,
- [ ] abort any ongoing transfer in the DMA controller,
- [ ] purge any queued DMA_COMPLETE callback in the callback queue.

This should result in a clean state and no dangling DMA interrupts.

One point to note is that the function can _not_ distinguish who requested the queued DMAs. It's all or nothing. If the application has more that one source of DMA requests then calling this function may deadlock some of the requesting processes.

alan-stokes, 2016-07-21:
having discussed via email. I think this is a suitable adjustment. Question is when people will get to do it. The EIEIO2 interface is getting rather late for going in the next release. Is it fair to assume it'll be ready for the release after this one. so about April 2017???

lplana, 2020-05-06:
The function requested in this issue was implemented in pull request #103.

closed 2020-05-06 by lplana