SpiNNakerManchester/spinnaker_tools #108: Weird buffer corruption issue: possible cause in shm_send_msg()?

dkfellows, 2019-10-04:
**Background:** I'm hunting a problem that only happens rarely under _intense_ load (during fast data loading) where buffers end up getting corrupted. The symptom is that the buffer ends up with what appears to be contents that are partially formatted as for (a part of) the inbound data stream and partially formatted as for (a part of) the outbound data stream. Another symptom is that communications subsequent to this are intensely flaky.

**Possible Cause:** In `shm_send_msg()` (in `scamp/scamp-3.c`), we see this:
```c
uint shm_send_msg(uint dest, sdp_msg_t *msg) // Send msg AP
{
    vcpu_t *vcpu = sv_vcpu + dest;

    sdp_msg_t *shm_msg = sark_shmsg_get();
    if (shm_msg == NULL) {
        return RC_BUF;
    }

    sark_msg_cpy(shm_msg, msg);

    vcpu->mbox_ap_msg = shm_msg;
    vcpu->mbox_ap_cmd = SHM_MSG;

    sc[SC_SET_IRQ] = SC_CODE + (1 << v2p_map[dest]);

    volatile uchar flag = 0;
    event_t *e = event_new(proc_byte_set, (uint) &flag, 2);
    if (e == NULL) {
        sw_error(SW_OPT);
        sark_shmsg_free(shm_msg);
        return RC_BUF;          // !! not the right RC
    }

    uint id = e->ID;
    timer_schedule(e, 1000);    // !! const??

    while (vcpu->mbox_ap_cmd != SHM_IDLE && flag == 0) {
        continue;
    }

    if (flag != 0) {
        sark_shmsg_free(shm_msg);
        return RC_TIMEOUT;
    }

    timer_cancel(e, id);
    return RC_OK;
}
```
However, if a shared message is allocated but a timer event is not (which I'm not sure is impossible when SCAMP is heavily loaded) then the code path that activates the second `return RC_BUF;` occurs. We're not running in the mode where `sw_error()` will RTE, so the system will free the shared message buffer and leave it up to the caller to respond; since this is a one-way SDP message (because that's how the fast data in protocol works) the return value will be just dropped.

The problem? We've now freed the shared message for the next operation (and messages get reused rapidly because they're managed as a stack) despite the application core being signalled to handle it; it will continue to work with it and then will _also_ free it when it is done. This corrupts the shared message stack (which will severely degrade the overall system reliability) and potentially generates shared buffers with two cores writing into them at once.

If I'm right (please tell me if I'm wrong!) then the fix for the corruption might be to move the lines:
```c
    vcpu->mbox_ap_msg = shm_msg;
    vcpu->mbox_ap_cmd = SHM_MSG;

    sc[SC_SET_IRQ] = SC_CODE + (1 << v2p_map[dest]);
```
to below the point where we know that we have successfully allocated all resources to perform the message sending operation (i.e., to just above the `timer_schedule()` call).

Am I crazy in thinking that this might be the problem? Review of my reasoning is _extremely_ welcome!

rowleya, 2019-10-04:
It might not be *the* bug, but it certainly looks like *a* bug to me!

rowleya, 2019-10-04:
It might actually be even better to try to get the event first and then the shared message to avoid interacting with the locks if the local events are exhausted anyway.

PR #109 (2020-03-26): Allocate events before touching shared resources
Because shared resources are seen by other cores, it's best to acquire events (which are local) before doing anything shared (like allocating a message from the shared queue or touching a mailbox). Also, doing this means that the failure mode where a shared message is allocated but an
event is not (in `shm_send_msg`) can't result in a message getting used twice at once and can't corrupt the stack of shared messages.

Fixes #108

(This will need heavy testing, given that the problem was only occurring under very heavy load. Which is why nobody no [... cut]
commit c37589f7 (2020-03-26): Allocate events before touching shared resources

Because shared resources are seen by other cores, it's best to acquire
events (which are local) before doing anything shared (like allocating a
message from the shared queue or touching a mailbox). Also, doing this
means that the failure mode where a shared message is allocated but an
event is not (in shm_send_msg) can't result in a message getting used
twice at once and can't corrupt the stack of shared messages.

Fixes #108
closed 2020-03-26 by rowleya